iizukanao/picam #195: how to delete the picam v1.4.9 and install picam v2.0.9

rajeshkrarora1, 2023-01-16:
Hi iizukanao, Thankyou for such a wonderful project, Sir, my query is that I am using picam v1.4.9 along with nginx-rtmp-module to stream to youtube and vlc , but this does not work on raspberry pi 4, so i want to uninstall version 1.4.9 and install picam version 2.0.9 , please guide

Thanks 
Rajesh

iizukanao, 2023-01-17:
Hi Rajesh, thank you for using picam!
Please just replace the picam binary with the newer version and everything should work fine. Also make sure that libcamera is enabled instead of the legacy camera stack.

FYI, picam 1.4.9 works on Raspberry Pi 4 if the legacy camera stuck is enabled in raspi-config > Interface Options.

rajeshkrarora1, 2023-01-19:
Hi iizukanao, Thankyou so much for your prompt response, I am in a remote village from last two days, where my internet was not working so i could not see your answer, I will inform you to confirm after implementing your suggestions. Thankyou once again🙏🙏🙏🙏🙏 Rajesh

rajeshkrarora1, 2023-01-19:
Hi  iizukanao,  Your picam version 1.4.9 is based on buster and there is only legacy camera support till buster I suppose, please correct me if I am wrong🙏🙏

iizukanao, 2023-01-19:
You are correct. Bullseye is required to use libcamera.
- picam v1.4.9 is for legacy camera stack (Buster and Bullseye)
- picam v2.0.x is for libcamera (Bullseye)

rajeshkrarora1, 2023-01-19:
Thankyou iizukanao, but does that mean that version 1.4.9 cannot be used
with pi 4, as I am using the latest picam-streamer, which uses picam
version 1.4.9 and does not work on raspberry pi 4  ? 🙏🙏🙏🙏

iizukanao, 2023-01-19:
If you manually upgrade the OS and picam to the latest version, it should work on the Raspberry Pi 4.

rajeshkrarora1, 2023-01-20:
Ok Thanks Nao Iizuka🙏🙏🙏🙏