iizukanao/picam #196: Is it possible to record in a different resolution then the running stream simultaniosly

rajeshkrarora1, 2023-01-19:
Hi iizukanao, Is it possible to record the  A/V in different resolution from the running stream

iizukanao, 2023-01-19:
It's possible, but it requires a lot of CPU power to transcode the video.
To do so, start the ffmpeg receiver as follows.
`ffmpeg -i tcp://127.0.0.1:8181?listen <your desired audio/video parameters> <output_file>`

Next, run picam to send the stream to ffmpeg.
`picam --tcpout tcp://127.0.0.1:8181`

rajeshkrarora1, 2023-01-19:
Hi Iizukanao, I am using your picam-streamer, multiple streams can
accessed, via RTMP on port 1935, RTSP on port 80, HTTP on port 80. Is it
possible on picam-streamer to record with different parameters and stream
with different parameters simutaniously?

iizukanao, 2023-01-19:
Unfortunately, that is impossible.

rajeshkrarora1, 2023-01-20:
Ok, thanks for the reply.