iizukanao/picam #201: alsadev error: cannot set channel count for microphone (Invalid argument)

lt250r93, 2023-06-30:
Hi, I am having an issue running Picam . I am using a Samson Go Mic and a Raspberry Pi Zero W running Bullseye. When setting the ALSA device to hw:1,0 it returns the following: "error: cannot set channel count for microphone (Invalid argument)". How do I resolve this issue? 
      ~/picam $ sudo ./picam --alsadev hw:1,0
      error: cannot set channel count for microphone (Invalid argument)

If I run picam without alsadev it will run and start capturing, but will disable audio. 
        ~/picam $ sudo ./picam
        error: cannot open audio capture device 'hw:0,0': No such file or directory
        hint: specify correct ALSA device with '--alsadev <dev>'
        warning: audio capturing is disabled

Also get a warning about mismatch:
        WARN RPI raspberrypi.cpp:1357 Mismatch between Unicam and CamHelper for embedded data usage!

I did confirm that my microphone is hw:1,0.
        arecord -l
        **** List of CAPTURE Hardware Devices ****
        card 1: Mic [Samson Go Mic], device 0: USB Audio [USB Audio]
          Subdevices: 1/1
          Subdevice #0: subdevice #0

iizukanao, 2023-06-30:
Please try the following.
- Add `-c 1` option.
- Use `plughw:1,0` instead of `hw:1,0`.
- Create a virtual ALSA device that does downmix multiple audio channels to mono, and specify that virtual device with `--alsadev`.

lt250r93, 2023-06-30:
thanks `plughw:1,0` worked while having `-c 2` set.

closed 2023-07-01 by iizukanao

feacluster, 2023-11-26:
I have the same mic, Samson Go Mic . I believe it is a 2-channel stereo condenser mic. Initially I got garbled audio, but changing to plughw:1,0 worked. However, I get audio/video out of sync by one second. I tried adding -c 1 and -c 2 , but both gave the same garbled audio.  Is the "virtual ALSA device" the only solution? Or should I experiment with the sampling rate? I believe the mic is 44 khz?, but not sure.

feacluster, 2023-11-26:
I tried the virtual ALSA device by creating a file called .asoundrc in the main home directory as follows:

```
squash@court1:~ $ more .asoundrc
pcm.blaster {
  type plug
  slave {
    pcm "hw:1,0"
    channels 2
  }
  route_policy sum
}
```
I then called picam with: ~/picam/picam --alsadev blaster ....

I did a reboot but unfortunately still the same error.

feacluster, 2023-11-27:
I tried doing all the suggestion, but eventually what worked was using ffmpeg to fix it during the TS to MP4 conversion:

    ffmpeg -i $file -itsoffset -1.1 -i $file -map 0:0 -map 1:1 -c:v copy -c:a copy -bsf:a aac_adtstoasc test.mp4 > /dev/null 2>&1

For me 1.1 seconds seems to produce perfectly synced audio/video.

lt250r93, 2023-11-28:
Good to see you got it working. Is there a lag in your video and audio or was it a delay in your audio only?

feacluster, 2023-11-28:
> Good to see you got it working. Is there a lag in your video and audio or was it a delay in your audio only?

Not sure I follow your question, but I believe the audio was delayed. I would see something happen in the video then hear it a second later.

lt250r93, 2023-11-29:
> > Good to see you got it working. Is there a lag in your video and audio or was it a delay in your audio only?
> 
> Not sure I follow your question, but I believe the audio was delayed. I would see something happen in the video then hear it a second later.

I should have clarified in my previous comment since its not related to your issue.  With my setup there is a 5 to 8 second delay from when something moves or makes noise and it playing on VLC or equivalent player. I'm still working on minimizing the delay.

feacluster, 2024-05-03:
While my -itsoffset workaround works when converting the video. I can't get it to work with livestreaming on youtube.  Here is the command I am trying which does not generate any feed to youtube . It seems to hang on the mic input or something:

ffmpeg -i tcp://127.0.0.1:8181?listen -itsoffset -1.1 -i tcp://127.0.0.1:8181?listen -map 0:0 -map 1:1 -c:v copy -c:a aac -ar 44100 -ab 40000 -f flv rtmp://a.rtmp.youtube.com/live2/xxxxxxxx