iizukanao/picam #204: picam has stopped working after libcamera upgrade

david-kohen, 2023-08-10:
My picam has stopped working with this message: 
picam: error while loading shared libraries: libcamera.so.0.0.4: cannot open shared object file: No such file or directory

libcamera and libcamera-base are now at so.0.0.5

Am I correct in thinking that the only solution is to rebuild from scratch?

Thanks

david-kohen, 2023-08-10:
Ok - I just tried downloading the prebuilt image and that works fine - I can see that my picam had 0.0.4 hard coded and the prebuilt image uses libcamera.0.0 instead _ I'm guessing that this issue was caused by my doing something wrong when a I built picam - closing with thanks and apologies...

iizukanao, 2023-08-10:
@david-kohen You are right. I updated the binary version of picam without notice. Sorry for the confusion.