iizukanao/picam #205: ./picam: No such file or directory

kimiroo, 2023-09-09:
I'm having a problem starting up picam.

I followed the installation guide and also tried with freshly imaged pi.

```
$ ./picam --alsadev hw:1,0
-bash: ./picam: No such file or directory

$ ldd ~/picam/picam
        not a dynamic executable

$ ls -la
total 15448
drwxr-xr-x 3 pi pi     4096 Sep  9 23:55 .
drwxr-xr-x 5 pi pi     4096 Sep  9 23:52 ..
drwxr-xr-x 2 pi pi     4096 Sep  9 22:23 archive
lrwxrwxrwx 1 pi pi       14 Sep  9 22:23 hooks -> /run/shm/hooks
-rwxr-xr-x 1 pi pi 15803936 Sep  9 22:23 picam
lrwxrwxrwx 1 pi pi       12 Sep  9 22:23 rec -> /run/shm/rec
lrwxrwxrwx 1 pi pi       14 Sep  9 22:23 state -> /run/shm/state
```

I checked my current directory and file existence but it was there.

I'm currently using raspberry pi os lite (32bit) (bullseye).
Any ideas?

kimiroo, 2023-09-09:
It seems like picam is having problem importing libraries in 32bit varient of raspberry pi os.
Tried in 64bit raspberry pi os lite but it works like a charm.
However I need this to work in Pi zero.
Tried using sudo, installing libcamera but sudo didn't work and libcamera was up to date.

iizukanao, 2023-09-09:
Please use [picam-2.0.12-armv7l.tar.xz](https://github.com/iizukanao/picam/releases/download/v2.0.12/picam-2.0.12-armv7l.tar.xz) for 32-bit OS, though it has not been tested on Raspberry Pi Zero.

kimiroo, 2023-09-12:
Works perfectly. Thanks!

Though pi zero seems to be struggling with combining audio and video input. Perhaps zero 2w might be able to handle the job.

closed 2023-09-12 by kimiroo

iizukanao, 2023-09-14:
Thanks for the feedback. For older models like the Raspberry Pi Zero, try using a resolution of 1280x720 (`-w 1280 -h 720`) instead of the default 1920x1080.

kimiroo, 2023-09-14:
Yeah I tried lowering the resolution, but it seems to struggle with 720p too. If I don't use audio it can handle 720p. But when I enable audio, the audio sync drags and video chops once a while. Both in recording and streaming.
I haven't tested the recording since it wasn't my main objective, but the pi zero wasn't able to keep up when I streamed via rtmp. Only when I streamed at 480p it was able to stream smoothly without any sync drift or choopy fps.
I used RAM disk as mentioned in readme.md but the results were the same. The ffmpeg was using cpu almost 90%, and pi wasn't able to keep up. Perhaps with bit of ffmpeg and config tweaks 
it might be able to do 720p streaming with audio?

Also, the picam was displaying
`
data rate from microphone is too slow 
`
so changing the microphone might help?

iizukanao, 2023-09-19:
Sorry, I do not have a Raspberry Pi Zero so I cannot answer further on this issue. It would be better to use the Raspberry Pi Zero 2 W before purchasing another microphone.