iizukanao/picam #206: Rec folder duplication

Sdb-sign, 2023-11-05:
Hi, I am trying to use this with limited programming know how. I have managed to install and run it, but the rec folder is created with duplicates of the picam folder, and I am unable to delete the duplicates as they arent able to be found?

iizukanao, 2023-11-07:
Hello,

I'm trying to fully grasp the situation you're encountering. It seems like there might be unintentional duplicates of the 'picam' folder within the 'rec' directory. If these duplicates were created by mistake and you wish to remove them, you can use the `rm -rf` command in the terminal. This command will forcefully remove the specified directories and their contents.

Sdb-sign, 2023-11-21:
Hi, I managed to resolve this by deleting, rebooting and running the install process again.

closed 2023-11-22 by iizukanao