iizukanao/picam #207: unable to complete install

Jlafont1986, 2023-11-08:
hello, newbie here. just followed the install guide on the README and when I attempt to run it gives an error with libcamera.so.0.0: no such file or directory.
here is my attempt at install (note I did in fact update in a previous window).
any help is appreciated and please keep it simple for the newbie. thanks

parentals@babypi:~ $ sudo apt install libharfbuzz0b libfontconfig libepoxy0
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'libfontconfig1' instead of 'libfontconfig'
libharfbuzz0b is already the newest version (6.0.0+dfsg-3).
libfontconfig1 is already the newest version (2.14.1-4).
libepoxy0 is already the newest version (1.5.10-1).
The following package was automatically installed and is no longer required:
  libpisp0.0.1
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
parentals@babypi:~ $ sudo apt autoremove
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following packages will be REMOVED:
  libpisp0.0.1
0 upgraded, 0 newly installed, 1 to remove and 0 not upgraded.
After this operation, 543 kB disk space will be freed.
Do you want to continue? [Y/n] y
(Reading database ... 145140 files and directories currently installed.)
Removing libpisp0.0.1:arm64 (0.0.1-1) ...
Processing triggers for libc-bin (2.36-9+rpt2+deb12u3) ...
parentals@babypi:~ $ cat > make_dirs.sh <<'EOF'
#!/bin/bash
DEST_DIR=~/picam
SHM_DIR=/run/shm

mkdir -p $SHM_DIR/rec
mkdir -p $SHM_DIR/hooks
mkdir -p $SHM_DIR/state
mkdir -p $DEST_DIR/archive

ln -sfn $DEST_DIR/archive $SHM_DIR/rec/archive
ln -sfn $SHM_DIR/rec $DEST_DIR/rec
ln -sfn $SHM_DIR/hooks $DEST_DIR/hooks
ln -sfn $SHM_DIR/state $DEST_DIR/state
EOF
parentals@babypi:~ $ chmod +x make_dirs.sh
parentals@babypi:~ $ ./make_dirs.sh
parentals@babypi:~ $ wget https://github.com/iizukanao/picam/releases/download/v2.0.12/picam-2.0.12-`uname -m`.tar.xz
tar xvf picam-2.0.12-*.tar.xz
cp picam-2.0.12-*/picam ~/picam/
--2023-11-08 18:39:02--  https://github.com/iizukanao/picam/releases/download/v2.0.12/picam-2.0.12-aarch64.tar.xz
Resolving github.com (github.com)... 192.30.255.112
Connecting to github.com (github.com)|192.30.255.112|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://objects.githubusercontent.com/github-production-release-asset-2e65be/15804457/3344b14e-e381-42bd-9bce-62b9357de086?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20231108%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20231108T183902Z&X-Amz-Expires=300&X-Amz-Signature=ebf872dad2165fbbc1977a7db41f1b35cd37eae7e3791251e4c2803444348f3f&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=15804457&response-content-disposition=attachment%3B%20filename%3Dpicam-2.0.12-aarch64.tar.xz&response-content-type=application%2Foctet-stream [following]
--2023-11-08 18:39:02--  https://objects.githubusercontent.com/github-production-release-asset-2e65be/15804457/3344b14e-e381-42bd-9bce-62b9357de086?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20231108%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20231108T183902Z&X-Amz-Expires=300&X-Amz-Signature=ebf872dad2165fbbc1977a7db41f1b35cd37eae7e3791251e4c2803444348f3f&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=15804457&response-content-disposition=attachment%3B%20filename%3Dpicam-2.0.12-aarch64.tar.xz&response-content-type=application%2Foctet-stream
Resolving objects.githubusercontent.com (objects.githubusercontent.com)... 185.199.108.133, 185.199.110.133, 185.199.109.133, ...
Connecting to objects.githubusercontent.com (objects.githubusercontent.com)|185.199.108.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 6119276 (5.8M) [application/octet-stream]
Saving to: ‘picam-2.0.12-aarch64.tar.xz’

picam-2.0.12-aarch6 100%[===================>]   5.83M  2.21MB/s    in 2.6s    

2023-11-08 18:39:05 (2.21 MB/s) - ‘picam-2.0.12-aarch64.tar.xz’ saved [6119276/6119276]

picam-2.0.12-aarch64/
picam-2.0.12-aarch64/picam
picam-2.0.12-aarch64/LICENSE
parentals@babypi:~ $ cd ~/picam
./picam --alsadev hw:1,0
./picam: error while loading shared libraries: libcamera.so.0.0: cannot open shared object file: No such file or directory
parentals@babypi:~/picam $

iizukanao, 2023-11-09:
Hello,
What version of the Raspberry Pi OS are you using?
You can check the OS version with the following command.
```
cat /etc/os-release
```

Jlafont1986, 2023-11-09:
Realized I need to use Bullseye. So I newly installed bullseye and this is what I got trying to install Picam:

parentals@babypi:~ $ sudo apt install libharfbuzz0b libfontconfig libepoxy0
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'libfontconfig1' instead of 'libfontconfig'
libepoxy0 is already the newest version (1.5.5-1).
libepoxy0 set to manually installed.
libfontconfig1 is already the newest version (2.13.1-4.2).
libfontconfig1 set to manually installed.
libharfbuzz0b is already the newest version (2.7.4-1).
libharfbuzz0b set to manually installed.
The following package was automatically installed and is no longer required:
  libfuse2
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
parentals@babypi:~ $ cat > make_dirs.sh <<'EOF'
#!/bin/bash
DEST_DIR=~/picam
SHM_DIR=/run/shm

mkdir -p $SHM_DIR/rec
mkdir -p $SHM_DIR/hooks
mkdir -p $SHM_DIR/state
mkdir -p $DEST_DIR/archive

ln -sfn $DEST_DIR/archive $SHM_DIR/rec/archive
ln -sfn $SHM_DIR/rec $DEST_DIR/rec
ln -sfn $SHM_DIR/hooks $DEST_DIR/hooks
ln -sfn $SHM_DIR/state $DEST_DIR/state
EOF

chmod +x make_dirs.sh
parentals@babypi:~ $ wget https://github.com/iizukanao/picam/releases/download/v2.0.12/picam-2.0.12-`uname -m`.tar.xz
tar xvf picam-2.0.12-*.tar.xz
cp picam-2.0.12-*/picam ~/picam/
--2023-11-09 18:20:50--  https://github.com/iizukanao/picam/releases/download/v2.0.12/picam-2.0.12-aarch64.tar.xz
Resolving github.com (github.com)... 192.30.255.113
Connecting to github.com (github.com)|192.30.255.113|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://objects.githubusercontent.com/github-production-release-asset-2e65be/15804457/3344b14e-e381-42bd-9bce-62b9357de086?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20231109%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20231109T182051Z&X-Amz-Expires=300&X-Amz-Signature=a5554f76757442af8a4c1b1551fdf634223e5d04e181d82c918b138b769b2bd2&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=15804457&response-content-disposition=attachment%3B%20filename%3Dpicam-2.0.12-aarch64.tar.xz&response-content-type=application%2Foctet-stream [following]
--2023-11-09 18:20:51--  https://objects.githubusercontent.com/github-production-release-asset-2e65be/15804457/3344b14e-e381-42bd-9bce-62b9357de086?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20231109%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20231109T182051Z&X-Amz-Expires=300&X-Amz-Signature=a5554f76757442af8a4c1b1551fdf634223e5d04e181d82c918b138b769b2bd2&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=15804457&response-content-disposition=attachment%3B%20filename%3Dpicam-2.0.12-aarch64.tar.xz&response-content-type=application%2Foctet-stream
Resolving objects.githubusercontent.com (objects.githubusercontent.com)... 185.199.111.133, 185.199.109.133, 185.199.108.133, ...
Connecting to objects.githubusercontent.com (objects.githubusercontent.com)|185.199.111.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 6119276 (5.8M) [application/octet-stream]
Saving to: ‘picam-2.0.12-aarch64.tar.xz’

picam-2.0.12-aarch6 100%[===================>]   5.83M  1.53MB/s    in 4.1s    

2023-11-09 18:20:55 (1.42 MB/s) - ‘picam-2.0.12-aarch64.tar.xz’ saved [6119276/6119276]

picam-2.0.12-aarch64/
picam-2.0.12-aarch64/picam
picam-2.0.12-aarch64/LICENSE
cp: cannot create regular file '/home/parentals/picam/': Not a directory
parentals@babypi:~ $

iizukanao, 2023-11-10:
It seems like you are almost successful. What were you doing when you encountered the following error?
> cp: cannot create regular file '/home/parentals/picam/': Not a directory

The main program is picam-2.0.12-aarch64/picam, so please move to the `picam-2.0.12-aarch64` directory and try running `picam`. You can do this by entering the following commands:

```bash
cd picam-2.0.12-aarch64
./picam --help
```

iizukanao, 2023-11-10:
If you are still interested in running picam on bookworm, I have uploaded the binary for bookworm 64-bit.
https://github.com/iizukanao/picam/releases/download/v2.0.12/picam-2.0.12-aarch64.tar.xz

Jlafont1986, 2023-11-10:
ok thanks. if i continue to have problems here i will try that.

to answer your question, i was following the install instructions when that error came up. specifically, it was the install step of the program. here is what i get when i try to move to the directory as you suggested:

parentals@babypi:~ $ cd picam-2.0.12-aarch64/picam
bash: cd: picam-2.0.12-aarch64/picam: Not a directory
parentals@babypi:~ $

iizukanao, 2023-11-10:
I see. That should be `cp`, not `cd`.
The complete command is `cp picam-2.0.12-*/picam ~/picam/`.

Jlafont1986, 2023-11-10:
just tried the new command and got this

parentals@babypi:~ $ cd picam-2.0.12-aarch64/picam
bash: cd: picam-2.0.12-aarch64/picam: Not a directory
parentals@babypi:~ $ cp picam-2.0.12-*/picam ~/picam/
cp: cannot create regular file '/home/parentals/picam/': Not a directory
parentals@babypi:~ $

iizukanao, 2023-11-10:
Maybe you have not created the directory `/home/parentals/picam/` yet.
Please follow the [Using a binary release](https://github.com/iizukanao/picam#using-a-binary-release) instructions again. `./make_dirs.sh` will create the directory `/home/parentals/picam/`.

Jlafont1986, 2023-11-10:
ok i will try that. thanks

Jlafont1986, 2023-11-10:
it looks like everything installed properly but when i try to run this is what i get:

parentals@babypi:~ $ cd ~/picam
./picam --alsadev hw:1,0
bash: ./picam: No such file or directory
parentals@babypi:~/picam $

iizukanao, 2023-11-10:
Looks like picam is still not installed properly. I think you entered wrong command during the installation.
What is the output of `ls -lR ~/picam`?

Jlafont1986, 2023-11-10:
i was copying and pasting to try to avoid any problems... thanks for having some patience with novice like myself. when i run i get this

parentals@babypi:~ $ ls -lR ~/picam
/home/parentals/picam:
total 15440
drwxr-xr-x 2 parentals parentals     4096 Nov 10 14:25 archive
lrwxrwxrwx 1 parentals parentals       14 Nov 10 14:25 hooks -> /run/shm/hooks
-rwxr-xr-x 1 parentals parentals 15803936 Nov 10 14:26 picam
lrwxrwxrwx 1 parentals parentals       12 Nov 10 14:25 rec -> /run/shm/rec
lrwxrwxrwx 1 parentals parentals       14 Nov 10 14:25 state -> /run/shm/state

/home/parentals/picam/archive:
total 0
parentals@babypi:~ $

iizukanao, 2023-11-10:
No problem. The files and directories seem to be set up correctly.
Can you run the commands one by one? First, enter `cd ~/picam`:
```
parentals@babypi:~ $ cd ~/picam
```
Then the shell prompt will change to `parentals@babypi:~/picam`, indicating that you have moved to the `~/picam` directory. Now, enter `./picam --alsadev hw:1,0`.
```
parentals@babypi:~/picam $ ./picam --alsadev hw:1,0
```

Jlafont1986, 2023-11-10:
heres the result

parentals@babypi:~ $ cd ~/picam
parentals@babypi:~/picam $ ./picam --alsadev hw:1,0
bash: ./picam: No such file or directory
parentals@babypi:~/picam $

iizukanao, 2023-11-10:
I'm sorry, but I cannot understand what is happening in your environment. Instead of the steps you've taken so far, please try running picam-2.0.12-aarch64/picam directly after extracting the downloaded file. If you're unsure how to do this, I recommend learning the basics of the Linux shell first.

Jlafont1986, 2023-11-11:
ok so ive re installed bullseye (64 bit) and reinstalled picam. when i try to run picam this is what i get:

parentals@babypi:~ $ cd picam
parentals@babypi:~/picam $ ./picam --alsadev hw:1,0
./picam: error while loading shared libraries: libcamera.so.0.1: cannot open shared object file: No such file or directory

im not sure why is says there is no file for libcamera, i just used it successfully. here is libcamera-hello command:
parentals@babypi:~ $ libcamera-hello
Made X/EGL preview window
[2:14:32.564694443] [2392]  INFO Camera camera_manager.cpp:297 libcamera v0.0.5+83-bde9b04f
[2:14:32.633202557] [2393]  WARN RPI vc4.cpp:383 Mismatch between Unicam and CamHelper for embedded data usage!
[2:14:32.634742371] [2393]  INFO RPI vc4.cpp:437 Registered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media2 and ISP device /dev/media0
[2:14:32.634858389] [2393]  INFO RPI pipeline_base.cpp:1101 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
[2:14:32.638525054] [2392]  INFO Camera camera.cpp:1033 configuring streams: (0) 1640x1232-YUV420
[2:14:32.639851109] [2393]  INFO RPI vc4.cpp:565 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 1640x1232-SBGGR10_1X10 - Selected unicam format: 1640x1232-pBAA

Any suggestions appreciated ;)

iizukanao, 2023-11-11:
For bullseye 64 bit, download this instead.
https://github.com/iizukanao/picam/releases/download/v2.0.12/picam-2.0.12-bullseye-aarch64.tar.xz

Jlafont1986, 2023-11-11:
looks like that did something :)
However, now the microphone is a problem. 
parentals@babypi:~ $ cd ~/picam
./picam --alsadev hw:1,0
error: cannot set channel count for microphone (Invalid argument)
parentals@babypi:~/picam $ arecord -l
**** List of CAPTURE Hardware Devices ****
card 1: Mic [Samson Go Mic], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

when i run without the mic attached i get this:
parentals@babypi:~/picam $ ./picam --alsadev hw:1,0.
ALSA lib conf.c:5112:(parse_args) Parameter DEV must be an integer
ALSA lib conf.c:5217:(snd_config_expand) Parse arguments error: Invalid argument
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM hw:1,0.
error: cannot open audio capture device 'hw:1,0.': Invalid argument
hint: specify correct ALSA device with '--alsadev <dev>'
warning: audio capturing is disabled
[0:56:01.804064489] [2158]  INFO Camera camera_manager.cpp:297 libcamera v0.0.5+83-bde9b04f
[0:56:01.846214324] [2160]  WARN RPI vc4.cpp:383 Mismatch between Unicam and CamHelper for embedded data usage!
[0:56:01.847373897] [2160]  INFO RPI vc4.cpp:437 Registered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media4 and ISP device /dev/media0
[0:56:01.848475803] [2158]  INFO Camera camera.cpp:1033 configuring streams: (0) 1920x1080-YUV420
[0:56:01.849268080] [2160]  INFO RPI vc4.cpp:565 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 1920x1080-SBGGR10_1X10 - Selected unicam format: 1920x1080-pBAA
capturing started

iizukanao, 2023-11-12:
Don't include the `.` at the end of `hw:1,0`
`./picam --alsadev hw:1,0`

Jlafont1986, 2023-11-12:
Didn't appear to be the issue...

parentals@babypi:~ $ cd picam
parentals@babypi:~/picam $ ./picam --alsadev hw:1,0
created directory: ./rec/tmp
error: cannot set channel count for microphone (Invalid argument)
parentals@babypi:~/picam $ arecord -l
**** List of CAPTURE Hardware Devices ****
card 1: Mic [Samson Go Mic], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

iizukanao, 2023-11-14:
Try using `plughw:1,0` instead of `hw:1,0`

Jlafont1986, 2023-11-18:
That was it. Thanks a bunch for all help!

closed 2023-11-18 by Jlafont1986

iizukanao, 2023-11-19:
I'm glad I could help! Happy to hear the issue is resolved.