iizukanao/picam #208: V4L2 H264 Encoder error

Sdb-sign, 2023-11-21:
Hi, 

I am trying to use Picam on an Rpi 5 with v3 camera module. The libcamera-hello works fine, but I get the following error when trying to run picam:

:~/picam $ sudo ./picam --alsadev hw:0,0
created directory: ./rec/tmp
[0:10:05.129193900] [2258]  INFO Camera camera_manager.cpp:284 libcamera v0.1.0+99-4a23664b
[0:10:05.145681168] [2263]  INFO RPI pisp.cpp:652 libpisp version v1.0.0 ce5624ed8726 23-10-2023 (07:46:39)
[0:10:05.167222421] [2263]  INFO RPI pisp.cpp:1106 Registered camera /base/axi/pcie@120000/rp1/i2c@80000/imx708@1a to CFE device /dev/media1 and ISP device /dev/media2 using PiSP variant BCM2712_C0
[0:10:05.167751129] [2258]  WARN V4L2 v4l2_pixelformat.cpp:336 Unsupported V4L2 pixel format RPBP
[0:10:05.168185889] [2258]  INFO Camera camera.cpp:1181 configuring streams: (0) 1920x1080-YUV420
[0:10:05.168527016] [2263]  INFO RPI pisp.cpp:1390 Sensor: /base/axi/pcie@120000/rp1/i2c@80000/imx708@1a - Selected sensor format: 2304x1296-SBGGR10_1X10 - Selected CFE format: 2304x1296-PC1B
ERROR: *** failed to open V4L2 H264 encoder ***

iizukanao, 2023-11-22:
Raspberry Pi 5 is not yet available in my country, so I don't think I can fix it right away.

Humpfldump, 2024-07-01:
Is there any update on the RPi5 support?

Humpfldump, 2024-12-01:
> Raspberry Pi 5 is not yet available in my country, so I don't think I can fix it right away.

I saw you updated the Readme, that the RPi5 is not supported. Is this a permanent issue or something which can be fixed?
If it's the latter, could you be so kind and provide information on what's the issue? I'm really no expert at the encoding stuff, but I can give it a try and have a look if you could kindly provide some information on what the issue is.

iizukanao, 2024-12-02:
@Humpfldump 
Thank you for reaching out and noticing the readme update. My apologies for not replying to your previous message—I appreciate your patience.

The lack of Raspberry Pi 5 support is not a permanent issue, but currently I don't have the bandwidth to add Raspberry Pi 5 support. Unlike previous models, the Raspberry Pi 5 lacks an H.264 hardware encoder, so software encoding using the FFmpeg library (libavcodec) is required. I believe this is the main step needed to make the Raspberry Pi 5 compatible.

If you'd like to explore it, I'd be happy to provide guidance on the codebase and relevant changes. Thanks again for your interest and offer to help—it means a lot!

Humpfldump, 2024-12-02:
@iizukanao
No worries about the delay, your effort on Picam is greatly appreciated and it's just normal that there are times, when you don't have as much time to spend on a project as before. So thank you for still sticking around :)

The past few days I was doing some research on what the reason for the error message could be and already came across another post hinting to recompile FFmpeg with only software encoders, instead of an H.264 HW encoder. Would that be a starting point for me?
Again, I'm not sure of how much help I can be - I'm a developer, but completely lacking background with media encoding and stuff around it. ;) But if you have a minute, I'm happy to give it a try, when I have some spare minutes.

iizukanao, 2024-12-04:
@Humpfldump Hello, thank you for your kind message and I sincerely appreciate your support and interest in picam.

You do not need to have in-depth knowledge of media encoding to contribute to this feature. The main task is to pass the data to ffmpeg's libav with the correct parameters and receive the result.

Instead of recompiling ffmpeg, we need to dynamically switch between H.264 software encoding (using libav) and hardware encoding, depending on whether picam is running on a Raspberry Pi 5 or not.

For the implementation, I refer to [rpicam-apps](https://github.com/raspberrypi/rpicam-apps) and use it as a guide on how to implement the encoding. In picam, [video_encoder.cpp](https://github.com/iizukanao/picam/blob/master/video_encoder/video_encoder.cpp) contains code for H.264 hardware encoding, and this is the counterpart to rpicam-apps [h264_encoder.cpp](https://github.com/raspberrypi/rpicam-apps/blob/main/encoder/h264_encoder.cpp). rpicam-apps also has [libav_encoder.cpp](https://github.com/raspberrypi/rpicam-apps/blob/main/encoder/libav_encoder.cpp) for H.264 software encoding, and picam needs a counterpart to this.

Let me know if you'd like further clarification or assistance!

Humpfldump, 2024-12-09:
@iizukanao Hello again and thanks a lot for your quick introduction. I'll try to have a look at it as soon as I can spare a few minutes - but to be honest, I cannot provide a schedule for it, but I'll try. ;)