iizukanao/picam #210: I cannot start recording

nahuel915, 2023-12-28:
Hello, first of all thank you for doing this great job!

Since a couple of days I tried to solve all the issues. Unfortunately, I have the following problem:
After installing bullseye I followed all the steps and I run picam with my ALSA device name it says "capturing started". But when I run "$ touch hooks/start_record" nothing happens. I took a look in the corresponding folder and there is not file created. 

Basically, I want to stream audio and video, so recording is not so important for but I think there is something wrong. 

Maybe I am missing something, It would be great if someone can help me.

Thank you!

iizukanao, 2023-12-28:
Thank you for your message.
To investigate the issue, try running picam with the `--verbose` option. If `touch hooks/start_record` command is accepted correctly, you will see a message like `start rec to` in the picam's verbose output.

nahuel915, 2023-12-29:
Thank you iizukanao, sorry if I am doing dumb mistakes I am trying my best.

Below is what I get after the command: pi@babypi:~/picam $ ./picam --alsadev plughw:2,0 -c 2 --verbose

video_width=1920
video_height=1080
video_fps=30.000000
video_pts_step=3000
video_gop_size=30
video_hflip=0
video_vflip=0
video_bitrate=3000000
video_avc_profile=high
video_avc_level=4.1
video_qp_min=-1
video_qp_max=-1
video_qp_initial=-1
video_slice_dquant=-1
video_brightness=0
video_contrast=0
video_saturation=0
video_sharpness=0
video_autofocus_mode=continuous
video_lens_position=-1.000000
alsa_dev=plughw:2,0
audio_channels=2
audio_sample_rate=48000
audio_bitrate=40000
audio_volume_multiply=1.000000
ng_thresh_volume=1.000000
ng_attack_time=0.200000
ng_hold_time=1.000000
ng_release_time=0.500000
is_hlsout_enabled=0
is_hls_encryption_enabled=0
hls_keyframes_per_segment=1
hls_number_of_segments=3
hls_encryption_key_uri=stream.key
hls_encryption_key=0x75b0a81de17487c88a47507a7e1fdf73
hls_encryption_iv=0x000102030405060708090a0b0c0d0e0f
hls_output_dir=/run/shm/video
rtsp_enabled=0
rtsp_video_control_path=/tmp/node_rtsp_rtmp_videoControl
rtsp_audio_control_path=/tmp/node_rtsp_rtmp_audioControl
rtsp_video_data_path=/tmp/node_rtsp_rtmp_videoData
rtsp_audio_data_path=/tmp/node_rtsp_rtmp_audioData
tcp_enabled=0
tcp_output_dest=
auto_exposure_enabled=0
auto_exposure_threshold=5.000000
is_vfr_enabled=0
white_balance=auto
exposure_control=auto
awb_red_gain=0.000000
awb_blue_gain=0.000000
metering=average
manual_exposure_compensation=0
exposure_compensation=0.000000
manual_exposure_aperture=0
exposure_aperture=0
manual_exposure_shutter_speed=0
exposure_shutter_speed=0
manual_exposure_sensitivity=0
exposure_sensitivity=0
roi_left=0.000000
roi_top=0.000000
roi_width=1.000000
roi_height=1.000000
min_fps=-1.000000
max_fps=-1.000000
[... 140 lines cut ...]
.............vSPEED_UP(video_pts_step=3000 audio_pts_step=1920 pts_diff=10320)................. 28.76 fps k=3 a-v=9690 c-a=-25070 u=1 d=0 pts=297510
.............................. 28.79 fps k=4 a-v=9270 c-a=-25356 u=1 d=0 pts=392010
.............................. 28.75 fps k=5 a-v=8850 c-a=-25493 u=1 d=0 pts=486510
.............................. 28.89 fps k=6 a-v=8430 c-a=-26063 u=1 d=0 pts=581010
.............................. 28.99 fps k=7 a-v=6090 c-a=-25064 u=1 d=0 pts=675510
.............................. 28.94 fps k=8 a-v=5670 c-a=-25788 u=1 d=0 pts=770010
........................vNORMAL...... 29.19 fps k=9 a-v=4230 c-a=-25402 u=1 d=0 pts=863610
.............................. 28.89 fps k=10 a-v=8310 c-a=-25977 u=1 d=0 pts=953610
......................vSPEED_UP(video_pts_step=3000 audio_pts_step=1920 pts_diff=10350)........ 28.81 fps k=11 a-v=9270 c-a=-24363 u=2 d=0 pts=1044810
.........disk_usage=13% start rec to rec/archive/2023-12-29_12-31-42.ts
...............caught up...... 28.82 fps k=12 a-v=8850 c-a=-24745 u=2 d=0 pts=1139310
.............................. 29.09 fps k=13 a-v=8430 c-a=-25991 u=2 d=0 pts=1233810
.............................. 28.85 fps k=14 a-v=6090 c-a=-24540 u=2 d=0 pts=1328310
.............................. 28.78 fps k=15 a-v=5670 c-a=-24797 u=2 d=0 pts=1422810
.............................. 29.07 fps k=16 a-v=5250 c-a=-25980 u=2 d=0 pts=1517310

While the command is running it keeps showing this fps data. You said to try the command "touch hooks/start_record" but the process is running and I did not know how to run that, so I opened a new terminal and run in /picam (while the other terminal was running) and nothing happened.

I had to put the modified command for my microphone "plughw:2,0 -c 2" as I read in one of the comments. 

So I tried the following steps to see how far I can get. In the part of Usage I did not do the steps "_Create symbolic links (optional, but strongly recommended)_". I did the steps for "_HTTP Live Streaming (HLS)_" and it is working but with the 4 seconds lag that is normal for this method. Then I tried to configure the "_Using picam in combination with nginx-rtmp-module_" I think I configured everything okay but is not running and I read in one of the comments that if the "touch hooks/start_record" is not working this might be an issue so I am back to 0.

PD: sometimes when I try again to run the picam command I get the error: ./state is a broken symbolic link. But this I solve it doing all the steps again from "# Create directories and symbolic links".

Thank you for your patience.

nahuel915, 2023-12-29:
[Update]

Sorry I do not know if I can continue with something related in this thread. 

But I think I got the "_Using picam in combination with nginx-rtmp-module_" also running. I think because the video delay is exactly the same as HLS and the audio in both (HLS, rtmp) is impossible to understand is like the audio speed is 0.5x.
But the rtmp streaming I cannot see it on the web browser. I copy paste it in VLC (like for HLS). 
The command I used to run the rtmp is: ./picam --tcpout tcp://127.0.0.1:8181 --alsadev plughw:2,0 -c 2

I am new to this and I like to learn, sorry if I do something wrong.

Thank you again.

nahuel915, 2023-12-29:
[update]

I fix the audio speed doing the steps in the link below:

https://github.com/iizukanao/picam/issues/183#issuecomment-1353141207

But I still have the audio and video delay in both methods.

iizukanao, 2023-12-29:
Thanks for your feedback. The audio-video delay is a known limitation in the current picam version. I'm planning to introduce a command line option for adjusting this delay in a future update.

As for the recording, `touch hooks/start_record` worked successfully as indicated by the `start rec to rec/archive/2023-12-29_12-31-42.ts` message. Use `touch hooks/stop_record` to finalize the recording file located at rec/archive/2023-12-29_12-31-42.ts

Feel free to reach out if you have more questions or need further assistance.