iizukanao/picam #211: Installing Picam on RPI Zero 2

Jlafont1986, 2024-02-06:
Hello, 
Newbie having trouble installing Picam on rpi zero 2 W. I am running bullseye with desktop environment (at least until I get everything installed) and, apparently, don’t quite understand the installation instructions to get Picam installed. 
If someone would help me with what exactly the command for the wget at the “ install Picam binary” needs to be for bullseye-armv71, it’d be much appreciated. Normally, I don’t mind spending time figuring out something like this on my own, but the rpi zero with the monitor is so painfully slow it is simply draining too much of my time. 
Thanks in advance

iizukanao, 2024-02-08:
Please download this one!
https://github.com/iizukanao/picam/releases/download/v2.0.12/picam-2.0.12-bullseye-armv7l.tar.xz

Jlafont1986, 2024-02-20:
Thank you! Installed and working well!
I did run into an issue I can't seem to fix, maybe you can afford one more fix?

I am trying to run a script at boot. I have this script that works in a terminal just the way i want- sudo LIBCAMERA_RPI_TUNING_FILE=/usr/share/libcamera/ipa/rpi/vc4/imx219_noir.json ./picam -o /run/shm/hls -w 1640 -h 1232

I use an init.d script to run a bash script to launch picam. So far i can run this without issue-
sudo /home/parentalz/make_dirs.sh
sudo /home/parentalz/picam/picam -o /run/shm/hls -w 1640 -h 1232

The problem I am having is when i try to add the tuning file to the above bash script it doesnt get applied. any clue on how i can get the tuning file to work?
and again, thanks in advance

iizukanao, 2024-02-22:
@Jlafont1986 
Hello,
To address the issue with the tuning file not being applied, please export the environment variable as follows before running picam.
```
#!/bin/bash

# Set the tuning file environment variable
export LIBCAMERA_RPI_TUNING_FILE=/usr/share/libcamera/ipa/rpi/vc4/imx219_noir.json

# Now run picam with your options
/home/parentalz/picam/picam -o /run/shm/hls -w 1640 -h 1232
```

Jlafont1986, 2024-03-05:
Thank you for the reply.
Unfortunately, I couldn’t get it to work as written. It still pulls up Picam without utilizing the tuning file. Doublechecking sudo LIBCAMERA_RPI_TUNING_FILE=/usr/share/libcamera/ipa/rpi/vc4/imx219_noir.json ./picam -o /run/shm/hls -w 1640 -h 1232
still enables the tuning file to be used properly. Is there any other way? I’d really like this to run properly at boot up.
Thanks
-j

Jlafont1986, 2024-03-13:
UPDATE: I’ve tried registering the tuning file in .bashrc hoping it would adopt the tuning file but had no success. Running Libcamera-hello readily utilizes the tuning file but not when I run Picam, sadly. So far, the only way I can get Picam to use the tuning file is to cd into Picam and put the tuning file in before the ./Picam command. Again, I’m trying to get Picam to use the tuning file with a startup script- any help is appreciated.
Thanks.
-newbie

closed 2024-03-14 by Jlafont1986

iizukanao, 2024-03-14:
I apologize for not being able to help you. Did you solve the problem?

Jlafont1986, 2024-03-14:
Thank you for the reply.And yes I did. For whatever reason, to run Picam it needed a ‘sudo’. And after much research and troubleshooting, I learned ‘sudo’ resets the environment. After utilizing ‘sudo -E’ in your script it worked as hoped.Thanks for all the help.Sent from my overpriced tablet

iizukanao, 2024-03-15:
Ah, I see! I hadn't thought of the sudo issue. I'm glad you solved it!