iizukanao/picam #212: Building on buster: Error on `cmake ..`

ofirnachum, 2024-03-19:
I'm building from source on Raspbian buster. I got through most of the steps, but on the `cmake ..` step I get the following error:

```
Dynamic Range Compression (DRC) options :
                 from /home/ofirnachum/picam/preview/null_preview.cpp:12:                                                                 
/home/ofirnachum/picam/picam_option/picam_option.hpp:86:24: error: ‘AfModeEnum’ in namespace ‘libcamera::controls’ does not name a type   
   libcamera::controls::AfModeEnum af_mode;
                        ^~~~~~~~~~
compilation terminated due to -Wfatal-errors.
make[2]: *** [preview/CMakeFiles/preview.dir/build.make:63: preview/CMakeFiles/preview.dir/null_preview.cpp.o] Error 1                    
make[2]: *** Waiting for unfinished jobs....
Scanning dependencies of target libstate
[ 26%] Building C object libstate/CMakeFiles/libstate.dir/state.c.o
In file included from /home/ofirnachum/picam/muxer/muxer.hpp:6,
                 from /home/ofirnachum/picam/audio/audio.hpp:9,
                 from /home/ofirnachum/picam/audio/audio.cpp:8:
/home/ofirnachum/picam/picam_option/picam_option.hpp:86:24: error: ‘AfModeEnum’ in namespace ‘libcamera::controls’ does not name a type   
   libcamera::controls::AfModeEnum af_mode;
                        ^~~~~~~~~~
compilation terminated due to -Wfatal-errors.
make[2]: *** [audio/CMakeFiles/audio.dir/build.make:63: audio/CMakeFiles/audio.dir/audio.cpp.o] Error 1                                   
make[1]: *** [CMakeFiles/Makefile2:609: audio/CMakeFiles/audio.dir/all] Error 2                                                           
make[1]: *** Waiting for unfinished jobs....
In file included from /home/ofirnachum/picam/preview/preview.hpp:16,
                 from /home/ofirnachum/picam/preview/preview.cpp:12:
/home/ofirnachum/picam/picam_option/picam_option.hpp:86:24: error: ‘AfModeEnum’ in namespace ‘libcamera::controls’ does not name a type   
   libcamera::controls::AfModeEnum af_mode;
                        ^~~~~~~~~~
compilation terminated due to -Wfatal-errors.
make[2]: *** [preview/CMakeFiles/preview.dir/build.make:76: preview/CMakeFiles/preview.dir/preview.cpp.o] Error 1                         
make[1]: *** [CMakeFiles/Makefile2:418: preview/CMakeFiles/preview.dir/all] Error 2                                                       
[ 29%] Linking C static library libstate.a
[ 29%] Built target libstate
In file included from /home/ofirnachum/picam/muxer/muxer.hpp:6,
                 from /home/ofirnachum/picam/muxer/muxer.cpp:4:
/home/ofirnachum/picam/picam_option/picam_option.hpp:86:24: error: ‘AfModeEnum’ in namespace ‘libcamera::controls’ does not name a type   
   libcamera::controls::AfModeEnum af_mode;
                        ^~~~~~~~~~
compilation terminated due to -Wfatal-errors.
make[2]: *** [muxer/CMakeFiles/muxer.dir/build.make:63: muxer/CMakeFiles/muxer.dir/muxer.cpp.o] Error 1                                   
make[1]: *** [CMakeFiles/Makefile2:636: muxer/CMakeFiles/muxer.dir/all] Error 2                                                           
make: *** [Makefile:152: all] Error 2
```

ofirnachum, 2024-03-19:
Typo... this error appears on `make -j4` (after `cmake ...` finishes successfully)

iizukanao, 2024-03-25:
@ofirnachum Hi, thanks for your message! I pushed a commit that might fix the issue. Can you pull the latest code from the master branch and build it again?

ofirnachum, 2024-04-06:
Thanks! I was able to find a binary version that worked for me, but hopefully your fix will help someone in the future!

closed 2024-04-06 by ofirnachum