iizukanao/picam #214: Error running Picam: error while loading shared libraries

CMDS-DIRK, 2024-07-26:
Hi, trying out your project to set up a baby monitor but I ran into shared library issues. 

I have a pi 3b+ and flashed it with different raspbian OS's (bookwork, bullseye, buster) and also tried different versions of the project, but I keep running into this error:

This error is visible when running Picam 2.0.12 on raspbian bookworm 64bit (flashed with RPI Imager tool)

```
./picam: error while loading shared libraries: libcamera.so.0.2: cannot open shared object file: No such file or directory
```

If I try different versions of Picam, it gives me the same error but for different libcamera versions.

with 2.0.11 its:

```
 ./picam: error while loading shared libraries: libcamera.so.0.0.3: cannot open shared object file: No such file or directory
 ```
 
 On different raspbian versions I see that different versions of libcamera are installed but they never match.
 
 If I try to create a softlink to the existing version I get an icon error so that also doesnt seem to work.
 
 Should I try to build some specific version of libcamera that matches with the error? I just try it with fresh installs of the OS and picam without additional libraries, other than the ones mentioned in your guide.
 
 If I try the rpicam-vid commands they seem to work but really would like audio as well. 
 
 I have now bought a PI 5 because my PI 3 gave up on me yesterday. 
 
 Please advice.

iizukanao, 2024-07-27:
@CMDS-DIRK Hi, thank you for pointing that out!

I've updated the binary to address the library compatibility problems you've mentioned. Please download the updated binary from the following link: https://github.com/iizukanao/picam/releases/tag/v2.0.12

Please note that Raspberry Pi 5 does not have an H.264 hardware encoder, so picam does not currently support Raspberry Pi 5. If you wish to record video and audio simultaneously, you might be able to achieve similar results using only ffmpeg.

CMDS-DIRK, 2024-07-29:
Thanks for addressing. Can't test it at the moment, don't have an older Pi lying around will buy it later. If someone else encountered this, let me know via a comment and I'll close this one. 

Will look into ffmpeg, have been able to get streams running with WebRTC and MediaMTX, see last item [here](https://www.hackster.io/ievgenii-tkachenko/comparing-video-stream-latencies-raspberry-pi-5-camera-v3-6c0ee4)

And also via the [python module picamera2](https://github.com/raspberrypi/picamera2/blob/main/examples/mjpeg_server.py)

But those don't include sound.

ArturGasparyan, 2024-08-29:
Hello Mr. Iizuka, thank you for your nice project!

Unfortunately I have the next error of the same kind with Version 2.0.12. I'm using a Raspberry Pi Zero (armv6l but I'm using your armv7l version, because no alternative) on Rasbian Bullseye without Camera Legacy Support.

My error on run is

> ./picam: error while loading shared libraries: libcamera.so.0.3: cannot open shared object file: No such file or directory

(It's 0.3 not 0.0.3!)

On my Rasbian I find libcamera.so.0.0.5

So what I'm trying for the last couple of hours(!) is to find a way to install libcamera.so.0.3.0 via libcamera.org...

Maybe it would be easier to support the default for bullseye?

Also: on your readme-text there's still a section for bullseye with a special URL (-bullseye) and also respecting the "uname -m" parameter - it fails, because there is not URL found (404) with -bullseye or armv6l.

Thank you!

ArturGasparyan, 2024-08-31:
Nevermind, I started afresh with buster and it works.

iizukanao, 2024-09-15:
@ArturGasparyan Sorry for the confusion. The README has been corrected.

closed 2024-09-15 by iizukanao