iizukanao/picam #215: Command line arguments - only the first one is activated.

dickh768, 2024-11-28:
I have installed the binary version of picam on a Rpi3b  using 32bit Bookworm as the OS.   In the main, the program works very well, however as I try to tailor the audio and video to my requirements I find that only the first command line option takes effect and the remaining ones are ignored.  E.g.  " ./picam --alsadev pipewire –v 8000000 –f 25 –c 1 –r 16000 "   will successfully connect the Alsa input to the pipewire service but will not implement the remaining video and audio options.  Similarly, if I rearrange the order of the parameters, the first is correctly implemented but the rest are ignored.

My install uses the binary blob using 'wget https://github.com/iizukanao/picam/releases/download/v2.0.12/picam-2.0.12-`uname -m`.tar.gz'
My OS is Bookworm 32 with 'uname -a' returning:
Linux raspberrypi 6.6.51+rpt-rpi-v7 #1 SMP Raspbian 1:6.6.51-1+rpt3 (2024-10-08) armv7l GNU/Linux

Is there a workaround for this?

iizukanao, 2024-11-29:
Can you try replacing all the `–` (en dash) with `-` (hyphen)? Here's the corrected command for reference:
Alternatively, you can copy and paste this:

```
./picam --alsadev pipewire -v 8000000 -f 25 -c 1 -r 16000
```

Let me know if that works!

dickh768, 2024-11-29:
Thank you for the very quick reply.  

If I copy your text and paste it into the command line then everything works fine!!

The source of the problem seems to be that I document my development steps in a MS Word document.  My example command line will have been copied and pasted from Linux into Word - and then copied and pasted back into the Linux command line.  Yesterday when I pasted my MS Word string into Excel to examine the character codes, some of the '-' characters showed code 45 and some had code 150.  This morning, they are all showing as character 45!!  So I think there is some auto-formatting going on to 'improve' the visual appearance.

In any case I think you correctly identify that the problem relates to the ambiguity of the '-' symbol, so as long as I get the string right in my launch script I should be OK.

Thanks again

closed 2024-11-29 by dickh768

iizukanao, 2024-11-29:
You're welcome! Glad to hear the problem is resolved😊