iizukanao/picam #216: mpegts.c: sample_fmts is deprecated, -Werror in CMakeLists.txt

summrum, 2025-01-12:
Hi, I have just tried to build picam using ffmpeg.git on a Raspberry Pi 4 running 64 bit Raspberry Pi OS (up to date as of today, 12/01/2025). Ran in to the following error:
```
/home/summrum/picam/mpegts/mpegts.c: In function ‘is_sample_fmt_supported’: 
/home/summrum/picam/mpegts/mpegts.c:46:14: error: ‘sample_fmts’ is deprecated [-Werror=deprecated-declarations]
   46 |   const enum AVSampleFormat *p = codec->sample_fmts;
      |              ^~~~~~~~~~~~~~
compilation terminated due to -Wfatal-errors.
cc1: all warnings being treated as errors
```
Removing ```-Werror``` from CMakeLists.txt allows the build to proceed, so I have a working picam executable.

I had a look at the ffmpeg documentation to try and update the mpegts.c file and it looks like ```avcodec_get_supported_config()``` is the [replacement for getting codec capabilities](https://ffmpeg.org/doxygen/trunk/group__lavc__decoding.html#gadd58e6b0bbca99fdbc547efbaa6b0ef1); I tried modifying using the information given on that link and this [section of the avcodec source](https://ffmpeg.org/doxygen/trunk/avcodec_8c_source.html#l00800) but I don't understand C/C++ and every variation I have tried thus far leads to failed compilation.

Thank you.

iizukanao, 2025-05-03:
Thank you for the detailed report! The issue has been fixed in commit a8cf94031727dcc37af51d73698dae5adae62695.

closed 2025-05-03 by iizukanao

summrum, 2025-05-03:
Thank you for commenting to let me know, much appreciated!

commit a8cf9403 (2025-05-03): Use avcodec_get_supported_config()

AVCodec->sample_fmts has been deprecated

Fixes #216