iizukanao/picam #217: "symbol lookup error" and picam won't start

natriumlamp, 2025-02-19:
Hello.
picam simply fails to start with the error "symbol lookup error"

Detailed error message is like below:
picam: symbol lookup error: /lib/aarch64-linux-gnu/libcamera.so.0.3: undefined symbol: _ZN7libpisp22compute_optimal_strideER24pisp_image_format_config

My uname -a is like below:
Linux raspberrypi 6.6.31+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux

I guess this error is related with RPi's recent update with libcamera and/or libpisp but not sure.
Will provide more information if anyone is interested in investigating this issue.

natriumlamp, 2025-02-22:
Thank you for a quick fix!
I tested with a libcamera0.4 environment and the new version works.

closed 2025-02-22 by natriumlamp

iizukanao, 2025-02-22:
Thank you for confirming!