iizukanao/picam #220: Libcamera v0.5.0 issues

Bodizappa, 2025-05-01:
Trying to get Picam running on a fresh install of 32-bit Bookworm lite.

The default install results in Libcamera v0.3.0. After following the instructions for the current binary install of Picam, I receive the following error when trying to run ./picam:

`./picam: error while loading shared libraries: libcamera.so.0.4: cannot open shared object file: No such file or directory`

I then did apt update and apt upgrade, and the Libcamera version went to v0.5.0 (looks like this just hit recently). I get the same error when I run ./picam.....still looking for the v0.4.0 install.

`./picam: error while loading shared libraries: libcamera.so.0.4: cannot open shared object file: No such file or directory`

Is there a way to get around this? Do I need to install v0.4.0 manually?

Thanks...have been using this for awhile and really enjoy the SW. Trying to get new HW up and running from scratch and cant seem to get past this issue.

iizukanao, 2025-05-02:
Thanks for the heads-up! I'll upload a fixed binary compatible with libcamera v0.5 in a few days.

MrChuckFinley, 2025-05-02:
I was going to post the exact same issue. I've been using DietPi for a while (mainly for airplay 2 and raspotify), it doesn't come with libcamera installed so I had to run "apt install libcamera-apps-lite" first, oh and also had to install alsa for the mic. I wish Picam was included here https://github.com/MichaIng/DietPi/wiki/DietPi-Software-list
@iizukanao thanks a lot for your work!

iizukanao, 2025-05-03:
Binary release has been updated for libcamera 0.5.
https://github.com/iizukanao/picam/releases/tag/v2.0.12
For 32-bit, please use [picam-2.0.12-armv7l.tar.gz](https://github.com/iizukanao/picam/releases/download/v2.0.12/picam-2.0.12-armv7l.tar.gz)
If you encounter any issue, please let me know.

@MrChuckFinley Thanks for your message! I'll take a look at DietPi.

closed 2025-05-03 by iizukanao

Bodizappa, 2025-05-03:
@iizukanao Thank you! Will let you know how it works out.

MrChuckFinley, 2025-05-03:
@Bodizappa I'm interested in your results. I've just tried and got the following "error while loading shared libraries: libX11.so.6: cannot open shared object file: No such file or directory"
I'm pretty sure it's because I'm using DietPi. I don't know what to install to get those libraries.

iizukanao, 2025-05-04:
@MrChuckFinley libX11.so.6 is provided by the libx11-6 package. Please run `sudo apt install libx11-6` and see if this resolves the issue.

Bodizappa, 2025-05-05:
@MrChuckFinley @iizukanao 

Thank you for making the update for Libcamera 0.5.0. 

This seems to be working now on my hardware (Pi Zero 2W) using Pi OS Lite (32-bit) 2024-11-19 distro. 

As before, fresh install and Libcamera-hello --version returns the following:

rpicam-apps build: v1.5.2 a8ccf9f3cd9d 03-10-2024 (22:50:05)
libcamera build: v0.3.2+27-7330f29b

Ran apt update and apt upgrade and got the following:

rpicam-apps build: v1.7.0 5a3f5965aca9 30-04-2025 (11:55:14)
rpicam-apps capabilites: egl:0 qt:0 drm:1 libav:0
libcamera build: v0.5.0+59-d83ff0a4

Followed binary installation instructions for Picam per GitHub readme. Ran ./picam and....success! Capture, record, etc working again.

Will do more involved testing in a bit, but for now it seems good. Thanks again.

MrChuckFinley, 2025-05-06:
Hello,

@Bodizappa thanks for your feedback.

@iizukanao installing libx11-6 solved my issue indeed, thanks. I run into another issue (again because of dietpi I think): symlinks are not created after a reboot, but if I run the lines to create them, picam starts without issues or errors. I guess make_dirs.sh is not run at reboot.

I discovered Picam years ago here: https://kamranicus.com/building-a-raspberry-pi-3-baby-monitor/

The guide includes the creation of a script that runs Picam with parameters and the option to run the script at boot. I was wondering if the creation of the script could be included in the readme of the project. Thanks again!

iizukanao, 2025-05-27:
@feacluster Your libcamera version looks too old. The latest release of picam only supports libcamera 0.5 and 0.4. Try updating your system with `sudo apt update && sudo apt upgrade`
Let me know if it still doesn't work!

feacluster, 2025-05-27:
> [@feacluster](https://github.com/feacluster) Your libcamera version looks too old. The latest release of picam only supports libcamera 0.5 and 0.4. Try updating your system with `sudo apt update && sudo apt upgrade` Let me know if it still doesn't work!

I'm actually running bullseye, not bookworm. I did the full upgrade and that seems to update libcamera libraries to 0.0.5 . I have 13 other pies running bullseye and they are not updated recently and still use libcamera 0.0.4 . I tried creating a symbolic link from 0.0.4 to 0.0.5 but that didn't work..

feacluster, 2025-05-27:
Update: I tried to copy the libcamera.so.0.0.4 and libcamera-base.so.0.0.4 libraries from my older pis where bullseye has not been updated in a year or so. I also copied the picam executable that works on them. Now when I run picam ( after updating the LD_LIBRARY_PATH ) on the latest bullseye it says:

`[11:34:41.947833779] [17117] ^[[1;31mERROR ^[[1;37mIPAModule ^[[1;34mipa_module.cpp:312 ^[[0;32mipa_rpi_vc4.so: ^[[0mInvalid IPA module name 'rpi/vc4'
[11:34:41.948326497] [17117] ^[[1;32m INFO ^[[1;37mCamera ^[[1;34mcamera_manager.cpp:299 ^[[0mlibcamera v0.0.4+22-923f5d70
[11:34:41.967207273] [17120] ^[[1;31mERROR ^[[1;37mRPI ^[[1;34mraspberrypi.cpp:1352 ^[[0mFailed to load a suitable IPA library
[11:34:41.967339512] [17120] ^[[1;31mERROR ^[[1;37mRPI ^[[1;34mraspberrypi.cpp:1277 ^[[0mFailed to register camera imx219 10-0010: -22
ERROR: *** no cameras available ***
`

iizukanao, 2025-05-27:
@feacluster Unfortunately, the binary release for bullseye is no longer provided.
Please build picam from source by following the [instructions here](https://github.com/iizukanao/picam/blob/master/BUILDING.md).

Since the conversation has gone a bit off-topic, I'm going to lock this thread. Thanks for understanding!