iizukanao/picam #221: Unable to build ffmpeg from source on latest Bullseye

feacluster, 2025-05-27:
Unfortunately the precompiled picam binary no longer works on an updated Bullseye. So I am trying to build from source but run into these errors when building ffmpeg:

```
CC      libavcodec/aarch64/vp8dsp_init_aarch64.o
CC      libavcodec/aarch64/vp9dsp_init_10bpp_aarch64.o
CC      libavcodec/aarch64/vp9dsp_init_12bpp_aarch64.o
CC      libavcodec/aarch64/vp9dsp_init_aarch64.o
AS      libavcodec/aarch64/vp9mc_aarch64.o
libavcodec/aarch64/vp9mc_aarch64.S: Assembler messages:
libavcodec/aarch64/vp9mc_aarch64.S:30: Error: bad instruction `ldp x5,x6,[x2]'
libavcodec/aarch64/vp9mc_aarch64.S:31: Error: bad instruction `ldp x7,x8,[x2,#16]'
libavcodec/aarch64/vp9mc_aarch64.S:32: Error: bad instruction `stp x5,x6,[x0]'
libavcodec/aarch64/vp9mc_aarch64.S:33: Error: bad instruction `ldp x9,x10,[x2,#32]'
more snipped...
```

iizukanao, 2025-05-27:
I haven't seen this kind of error before, and honestly I'm not sure what's causing it, so I'm afraid I can't really help with this. Since it looks like an ffmpeg compilation issue rather than something specific to picam, maybe try reporting it to the ffmpeg project?

feacluster, 2025-05-28:
Thanks, but I found a workaround by just using libcamera-vid to record video and audio together . Here's the command I am using:

```libcamera-vid -t 10000 --width 1640 --height 1232 --bitrate 2000000 --framerate 30  --autofocus-mode default --codec libav --libav-audio --audio-source alsa --audio-device hw:$card,0 --audio-channels 1 --audio-bitrate 192000 --av-sync 200000 -n -o test.mp4 ```

If this proves to be unstable, then I will revisit building picam from source again. But I think many people will break their picam on bullseye if they do a full update/upgrade..

iizukanao, 2025-05-29:
Glad you found something that works for you.